QIDITECH/QIDISlicer #2: Issue for Mac Intel

Booldegum, 2023-08-24:
Hello, It does not work on Mac Intel Ventura 13.4.1. Could you help me please?

76Octane, 2023-08-26:
Hi @Booldegum ,
We apologize for the inconvenience caused.
Currently our software only supports Macs with M1 and M2 chips. Before the release of the M1 chip, Apple's MacBooks used Intel X86-based chips.
The problem you described is also the direction we are working on in the near future, and we will fix this problem as soon as possible.

QIDITECH, 2023-08-28:
We have optimized the compatibility of macos, please try to download and install again.
https://github.com/QIDITECH/QIDISlicer/releases/tag/V1.0.5

Booldegum, 2023-08-28:
Thank you, that was very quick ! The software has been correctly installed. There is no problem for the moment.
I can contact you if I find some issues if you want.

closed 2023-08-28 by Booldegum