QIDITECH/QIDISlicer #3: Label object doesn't work correctly

Lightning2X, 2023-08-24:
Label object always only produces a single object in GCODE output. The GCODE  file only includes the line `EXCLUDE_OBJECT_DEFINE NAME=stl_id_0_copy_0` Which creates an object around all the objects, instead of multiple objects.  Which results in fluidd only seeing one object instead of multiple

QIDITECH, 2023-08-26:
label objets now only supports octoprint, which is the original function of prusa, it does not support klipper's EXCLUDE_OBJECT. We will add this feature in subsequent updates, stay tuned.

QIDITECH, 2023-10-13:
We have added the ‘Exclude Objects’ in version 1.0.6, please download the latest version of the software via the following link.
https://github.com/QIDITECH/QIDISlicer/releases/tag/V1.0.6

closed 2023-10-13 by QIDITECH