QIDITECH/QIDISlicer #4: Infill bridging over supports.

r00tdownx, 2023-09-26:
Hi in new version of your slicer could you give us more options to control infill bridging over supports.
This is the only area I cannot get a clean surface, right where the print built off a support!
Thanks.
[image]

QIDITECH, 2023-09-27:
Maybe you can try to close Thick bridges.
[image]

r00tdownx, 2023-09-27:
Hi I'm still getting bad bridging over supports with thick bridges on.  Solid infill is above bridge infill.
[image]
[image]
![Screenshot_3](https://github.

I've attached my bridge test.  Maybe you can try and show results?
[Bridge test.zip](https://github.com/QIDITECH/QIDISlicer/files/12736223/Bridge.test.zip)
com/QIDITECH/QIDISlicer/assets/142807917/90f5d8cb-acea-45a6-8c32-a274775a23b3)

closed 2023-09-28 by QIDITECH