QIDITECH/QIDISlicer #6: Chamber temperature is not being set in the default printer G-code

stew675, 2023-10-21:
The default Machine G-code is current this:

```
G28
M141 S0
G0 Z50 F600
M190 S[first_layer_bed_temperature]
G28 Z
G29 ; mesh bed leveling ,comment this code to close it
G0 X0 Y0 Z50 F6000
M109 S[first_layer_temperature]
M83
G0 Z5 F1200
G0 X{first_layer_print_min[0]} Y{max(0, first_layer_print_min[1] - 2)} F12000
G0 Z0.2 F600
G1 E3 F1800
G0 Z0.3 F600
G1 X{min(first_layer_print_min[0] + 30,print_bed_max[0])} E6 F600
```

The `M141` macro is what instructs the printer to set the chamber temperature, and at present, it always sets it to 0
It seems to me that since the X-Plus3 and X-Max3 both have chamber heaters, then the chamber temperature as defined in the filament configuration should be respected.

This can be accomplished by changing the line starting with `M141` to the following

```
M141 S[chamber_temperature] ;set chamber temp
```

and that now instructs the printer to respect the chamber temperature settings in the filament specification

QIDITECH, 2023-10-24:
There is a bug in the chamber temperature setting in version 1.0.5, which will be fixed in the next version.

The chamber temperature will be activated after printing the first layer, because the first layer printing speed is slow, some manufacturers of ABS will be easy to clog.

closed 2023-10-24 by QIDITECH

braclark, 2023-12-08:
First, there is a setting in "Filament Settings" "Filament" "Temperature" "Chamber" "First Layer" for the user to set the chamber temperature for the first layer:
`[first_layer_volume_temperature]`
You are saying that you want the software to ignore what the user is setting that to because you know better then them?  Stratasys, the company that patented chamber heat, has always set the chamber temperature before printing. You should allow the user to change this, but could still set the default to 0.
`M141 S[first_layer_volume_temperature]` ;set chamber temp