QIDITECH/QIDISlicer #7: RFE: Support for i-Fast printer

donnydavis, 2023-10-25:
I have both an i-Fast and X-Smart 3. This software looks great, but it's less than ideal to use two slicers. 

Please add support for older model printers, they weren't cheap and are great machines.

QIDITECH, 2023-10-26:
We are trying to test and adjust the configuration of the printer, it may take some time to adapt to all models, you can contact our after-sales to get a simple version of the i-fast configuration file.
https://qidi3d.com/pages/warranty-policy-after-sales-support

closed 2023-10-26 by QIDITECH

donnydavis, 2023-10-28:
A simple version isn't really what I am after. 

Don't we just have to convert this code to make it work?

https://github.com/QIDITECH/Qidi-Print/blob/main/python/qidi/Wifi/CBDConnect.py

QIDITECH, 2023-10-30:
The assembly language used by the two software is different, and it is not a simple paste and copy, and the wifi function needs to be rewritten.
We need more time to solve these problems. Thank you for your support and understanding.

donnydavis, 2023-10-30:
Understandable, however the functions and logic are the same. No pressure on time, that isn't what I am trying to say.

QIDITECH, 2023-10-31:
Yes, all the code related to communication is in the following two links.
https://github.com/QIDITECH/Qidi-Print/tree/main/python/qidi/Wifi
https://github.com/QIDITECH/Qidi-Print/tree/main/plugins/ControlPanelPlugin