QIDITECH/QIDISlicer #8: pressure advance not resetting after print

embracemaking, 2023-10-28:
At some point this was added to filament start g-code:
; Filament gcode
{if (enable_advance_pressure[0] == true)}M900{else};M900{endif} K[advance_pressure] T[smooth_time]

Previously this if statement was not there. I have sliced files prior to pressure advance being written to start gcode. 

Now when I run a new file the pressure advance is set - but when I run an old file next the pressure advance values do not reset. Now when I print old gcode files in PLA the quality is ruined because it uses the pressure advance settings from the last job. In my opinion after a job is complete it should reset pressure advance or disable it so the next file can set it or not set it if it is an old file.

QIDITECH, 2023-10-28:
In v1.0.6, we fixed the bug of ’Pressure Advance‘ and moved it into the filament start code for easy user customization.

However, we have not changed the M900 generation logic, in older versions, M900 instructions will also be generated in all gcode unless you turn it off. And the older version did not restore the ’Pressure Advance‘ after the end of printing.

We recommend that you regenerate the gcode file or manually add the M900 to the old file to ensure that the correct pressure advance is enabled each time you print. We don't think it's a good idea to reset the pressure ahead in the end code, because there will be cases where you stop printing halfway and start printing another model again.

embracemaking, 2023-10-29:
Okay but something is still not right then... I just disabled pressure advance, re-sliced and I'm still getting M900 with K and T output in my gcode with the if statement present. I don't think it's executing properly.

closed 2023-10-30 by QIDITECH