QIDITECH/QIDISlicer #12: Can't send gcode on 1.0.8

asltpo, 2023-12-17:
Summary: since updating to 1.0.8 I can no longer send gcode directly to my printer over the network.

Expected behaviour (<1.0.8):
- Slice model
- Two buttons appear in the bottom right hand corner: "Export G-code" and "Export G-code and send to printer"
- The second button exports and transfers the file over the network, and the print begins immediately.
- Under the "Device" tab I can view the fluidd interface for my printer.  Once initiated this remains live even if I click away and come back.

Current behaviour (1.0.8):
- Slice model
- Only one button appears in the bottom right hand corner: "Export G-code".
- Every time I navigate to the "Device" tab I have to reselect my printer to view the fluidd interface.  It appears that the connection is lost/reset as soon as I click away.

My setup:
- Qidi X-Plus3 with firmware v4.2.10
- Qidi slicer 1.0.8
- Windows 11

QIDITECH, 2023-12-19:
We can't reproduce your problem, please pack your configuration folder and send it to me.
Or contact after-sales service for more timely service.

Cathobo, 2023-12-31:
Hi, I had the same problem and To get around it, I exported the g code file to my desktop. I found that the new 1.0.8 Klipper software was showing two identical QIDI machines on the same IP address. I deleted one then I connected the remaining so it would show on my device. 
I then, using the + Upload Button just bellow Jobs in the klipper terminal and selected the g code file I had exported to my desktop.

Cathobo, 2023-12-31:
I am running Windows 10, I forgot to mention one can also upload and Print using the + button in jobs.

QIDITECH, 2024-03-28:
This issue has been fixed in the new version, please update to the new version.

closed 2024-03-28 by QIDITECH