QIDITECH/QIDISlicer #13: Can't drag and drop stl or other fiels into the slicer

xTremePower, 2023-12-18:
Hello,

I own 28 FDM printers and use SuperSlicer, and now OrcaSlicer.

last week I brought 2 Qidi Tech X-Max 3, I love this printer a lot !!!
And because i'm lazy, I installed your slicer to obtain all printer profiles for Qidi, I see it's a fork of PrusaSlicer, it work great and I love the printer control interface.

But there was one annoying thing in your slicer : It's simply impossible to drag and drop any 3d files to import it in QidiSlicer, normally in all PrusaSlicer you can drag and drop an stl file into the software to make model import ultra easy. Why it's impossible in your slicer ? When I try, the mouse show a forbidden sign. It's a bug or it made on purpose ?

Thanks

PS : I think I will contact you to make some business together because I need to renew my FDM farm :p

jiweaver, 2023-12-18:
I've seen this problem before. I have not used the latest version of QIDI slicer (I've been away for a couple months) but I remember there was a way around it. It does work. But sometimes it would refuse to let me drop the files. Trying to remember... Are the files being dragged from within a compressed archive (like a zip file?) If so, try extracting it somewhere first. My memory is not good. But I think that was it.

xTremePower, 2023-12-18:
Hello,

I never drop from zip, because I only slice my own designs :p it dropped
from any computer folder but never compressed archive

QIDITECH, 2023-12-19:
We have not disabled this feature, it may be due to a permission issue or registry installation error. Try running the software as an administrator, or uninstall the software and reinstall it. If this does not work, please send me a video or picture so that I can see it more clearly, and provide the computer system version, hardware information, etc.

PS：If you want to get more cooperation, please contact us through after-sales service or the official website.
https://qidi3d.com/
https://qidi3d.com/pages/warranty-policy-after-sales-support

Phil1988, 2023-12-30:
on v1.0.7 I had this too.

on v1.0.8 this seems to be fixed now

Shrugsy, 2023-12-30:
I had this occurring briefly after updating to v1.0.8, but upon closing & re-opening QIDISlicer, I'm able to drag'n'drop STL files once again

QIDITECH, 2024-01-04:
Running the software immediately after the installation will cause this problem, and it can be used normally after restarting.

closed 2024-01-04 by QIDITECH