QIDITECH/QIDISlicer #18: Please allow other ports then 10088

Phil1988, 2023-12-30:
By default QidiSlicer sets the port to 10088.

If the 3D printer is on a different port, then it can connect to that printer.
[image]

Please allow to set a different port, so we can also use other printers with your slicer :)

QIDITECH, 2024-01-04:
Connecting printers from other manufacturers will cause errors in the multi-device management function, which we will solve in the subsequent optimization. It may not be realized in the short term, please understand.

whitekaren, 2024-03-21:
[image]
In the software version 1.1.2, we added Klipper(via QIDI) option to host type. When you select this option, the software will automatically add port 10088 when connecting. The original Klipper(via Moonraker) option no longer automatically adds port 10088, so you can select Klipper(via Moonraker) in version 1.1.2, and then add other ports in the url to connect to devices from other manufacturers, looking forward to your further feedback

closed 2024-03-21 by whitekaren

Phil1988, 2024-03-21:
Great work! 
Thanks a lot! 👍

Phil1988, 2025-01-25:
I get many user reports on this and tested it myself:

Its broken in QidiSlicer v1.2.1

[image]

Works in QidiSlicer v1.1.6:

[image]

Could you please have a look at it and fix it again?
Thanks!

Phil1988, 2025-02-27:
@QIDITECH Can you please have another look at this?