QIDITECH/QIDISlicer #19: Job Upload Date and Time Incorrect in QIDI Slicer

Cathobo, 2023-12-31:
I have had this problem with all versions of QIDI Slicer. In the Jobs Section under Modified the date is show as Oct 07 2023 and not Dec 31 2023. 
I contacted support and was given a terminal to correct the issue. It did but after turning my machine off and the rebooting the problem reappeared. 
The time zone on my pc is correct and the date and time format is correct in settings.

Phil1988, 2023-12-31:
Is this coming from the slicer or from the printer?
Please check the .gcode file if the date (Oct 07) is anywhere in it.

If NOT, then its coming from the printer.
Please update the time on your maschine as I mentioned here:
https://github.com/QIDITECH/QIDI_MAX3/issues/15#issuecomment-1863875835

Cathobo, 2023-12-31:
Thanks PhilThe date in the g code is correct. Support gave me a terminal called MobaXterm and a password and a file to flash the printer. Unfortunately it did not work. The printer is running at present . I will try your method .Neil

 
 

QIDITECH/QIDI_MAX3#15 (comment)

Cathobo, 2023-12-31:
I managed to connect to QIDI via my windows command prompt and changed the date using Sudo date, time and year, worked ok. I uploaded several gcode file and the time was showing correct. I then turned the QIDI off and waited a few minutes then tried again, all worked ok. I tried logging into my machine with root but was denied. I had to use makerbase.I then tried the sudo chmod command but came back no directory exists.I noticed when mks opened in my terminal the date last opened was Oct 7 2023, I will see what happens tomorrow. I think it will revert back to Oct 7.

Phil1988, 2024-01-01:
@Cathobo : just use the commands I directly linked. The `sudo date` will only temporaly work - untill you shut the printer off.

Where exactly doesnt it find the directory?
You may not have ntpd installed.

First: Use putty. (no cmd, no other software)
login as "mks" and use the password

use the 4 lines code I provided (one at a time).
If it fails at `sudo chmod -x /usr/sbin/ntpd` then ntp is not installed.

Then you have to insall it first. It will only work if your current time beeing correct, because apt-get needs correct time.
So set the date temporaly with `sudo date` and then do
`sudo apt-get install ntp`

After this is done, proceed with the given command lines I gave you :)

Cathobo, 2024-01-01:
Hi PhilYou are right the date and time was only temp.I have got ntp version 1:4.2.8p12+dfsg-4 installed.I used the sudo date again but when I used sudo chmod -x I get no such file or directory.Using sudo help I find -c -f -v and -R but no -x.Neil

Phil1988, 2024-01-03:
@Cathobo  It seem like you are not sure what to do and what you acually do :D 
`sudo help` is not a valid command.
sudo just gives super user access and "help" is not a command.

`sudo chmod` does "chmod" (change directory permissions) with super user access.
chmod does have a few parameters, one of it is -x (see here: https://www.pluralsight.com/blog/it-ops/linux-file-permissions).

You are still not giving information where your error happens and this is very frustrating for anyonw willing to help (like me).

You should use the commands I provided. On one of the two `sudo chmod`  you get an "no such file or directory"-error, but its still open on which one.
Is it on "usr/sbin/chronyd" or on "/usr/sbin/ntpd".

Please provide more information also with screenshots etc. otherwise noone can help you and has to guess riddles.

Cathobo, 2024-01-03:
Hi PhilThanks for your help, I have attached screenshots of the problem.. Screenshot 1: File gcode, Screenshot 2: QIDI Date settings , Screenshot 3: Uploaded g code file (wrong date), Screenshots 4 - 7  Putty terminal. Hope this helps.Neil

QIDITECH, 2024-01-04:
Thank you for your feedback, a small number of motherboards do have this problem, we are still investigating this problem.

Phil1988, 2024-01-05:
>  I have attached screenshots of the problem.. >
There is no screenshot attached here.
Please write directly on github, not as response via email.

look how your comments look like :)

Cathobo, 2024-01-05:
Hi Phil here are the screen shots I sent
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Phil1988, 2024-01-06:
Ahh ok, then you need to
`sudo apt-get install chrony`
first.

retry again please and tell us if it worked or what was the error :)

Cathobo, 2024-01-06:
Hi Phil,I installed chrony as you suggested and then entered the first line ok but now says cannot locate ntpd .
Neil

Phil1988, 2024-01-06:
Thats strange, because you said npt is already installed on your system...

you surely have done 
`sudo apt-get install ntp` ?

Cathobo, 2024-01-07:
[image]
Hi Phil, This is what happened after installing chron, it deleted ntp and says to install ntp, chron would be deleted.

Phil1988, 2024-01-07:
And you did
`sudo apt-get install chrony` ?

Cathobo, 2024-01-07:
[image]
Hi Phil Yes I did and it is still installed but npt is not. To install npt it would remove chron.
Neil

Phil1988, 2024-01-07:
contact qidi directly about that...
Om running out of ideas

Cathobo, 2024-01-08:
Thanks for your help Phil. An idea I have had is to run sudo date before I open the QIDI slicer, then the date would be correct. 
Regards Neil

Cathobo, 2024-01-08:
Hi PhilGreat news, I have just booted my QIDI and without using the terminal to set date and time, I uploaded a file and Voila! the date and time is correct. I have attached a screenshot. It must be using the chron app I installed yesterday without NTP.
Regards Neil.

Cathobo, 2024-01-08:
[image]
Here is the screen shot of correct time etc.

Cathobo, 2024-01-08:
Hi Again Phil,
If there is a way I can display my settings and screenshot them to you so we can help others with the problem let me know how to do it and hopefully resolve the issue,
Neil

closed 2024-02-22 by QIDITECH