QIDITECH/QIDISlicer #22: Filament profile selection

AstroKeith, 2024-01-07:
I expected to be able to select for printing one of the filaments 'ticked' in the 'add filament' tab. Maybe I'm missing something, but when I change filament types it seems I have to manually adjust all the parameters, surely the QIDI filaments are available as presets? If so how are they selected? 

I'm posting here as I have failed to find any detailed manuals or instructions on how to use QidiSlicer 1.0.8

QIDITECH, 2024-01-08:
When you switch consumables, a prompt box will pop up asking you whether to save the configuration.
You may have checked the Remember my choices option.
[image]

You can turn it back on in Preferences.
[image]

AstroKeith, 2024-01-08:
Thank you for your quick and detailed reply.
But the basic problem is - how do I switch consumables? Or select a Preset? It seems I can only edit things like nozzle temperature in the 'Filament Settings' Tab manually.
If I switch material to Qidi Rapido ABS from my usual PLA, is there not a standard preset I can select?

AstroKeith, 2024-01-08:
Ah, Solved
I found that you have to change the selected printer from the initially created one (astro3D in my case), to the "X-MAX 3 0.4 nozzle".
This is a great printer and the slicer seems powerful, but better instructions on its use are needed!

AstroKeith, 2024-01-08:
Closed. Thanks

closed 2024-01-08 by AstroKeith