QIDITECH/QIDISlicer #24: image on display whenprinting black

pmbroth, 2024-01-10:
Sending print jobs tp x-max 3, no image is diaplayed.  

it is set to Klipper  with  380x380/QIDI, 210x210/QIDI

pmbroth, 2024-01-10:
I have tried with several stl's  all do the same thing.  Screen image is black.  I rolled back to 1.08 and no issues,.  Re-installed 1.1 and still the same - black box.

QIDITECH, 2024-01-11:
Thank you very much for your testing and feedback, 1.1.0 has a lot of compatibility issues, we will continue to fix, and release again after completion.

pmbroth, 2024-01-12:
I removed the /qidi from the 380*380 seciton, and the thumbnail works wihtin the fluid display, but still does not show on the printer.

QIDITECH, 2024-01-13:
A little mistake, we have re-uploaded 1.1.0, please re-download and install. In addition, there may be other bugs, if found, please timely feedback to us.

closed 2024-01-13 by QIDITECH

pmbroth, 2024-01-13:
It Works!  Thank you Brian

 

Closed #24 <https://github.com/QIDITECH/QIDISlicer/issues/24>  as completed.