QIDITECH/QIDISlicer #28: V 1.1.0 geschweifte und eckige Klammer im Kommentar führt zum Fehler

derbasteller, 2024-01-19:
Es kommt oft vor, das ich Befehlzeilen auskommentieren möchte. Um bei Bedarf wieder zu aktivieren. Es gibt sehr viele Zeilen mit „[]“ eckigen und „{}“ geschweiften Klammern!
Diese führen dann, beim slicen, zu Fehlern und Abbruch.
Z.B.: M104 S{300-(9 * int((layer_z - 3) / 6))} funktioniert!
Die Beschreibung hierfür:
;M104 S{Starttemp–(Tempschritte*int((layer_z–HoeheBasis)/HoeheBereich))}
führt zur Fehlermeldung!
Benutzt habe ich diese im benutzerdefinierten Layerwechsel zum erstellen eines Temptower.

QIDITECH, 2024-01-22:
Custom Gcode must conform to format regulations.
What kind of gcode do you want to write?

derbasteller, 2024-01-22:
Chitu GCode, die auch im [before_layer_gcode} benutzt wird.

QIDITECH, 2024-01-23:
I'm sorry. Can you be more specific?
Chitu gcode also works, but some of the code will not work in the gcode preview.

closed 2024-02-22 by QIDITECH