QIDITECH/QIDISlicer #29: Qidislicer 1.1.0 on Linux

Vance1122, 2024-01-25:
When you add several files to the build plate and you use auto arrange then slice, The slicer complains that the models are too close and the tool head might hit other models while printing. It used to not do this in 1.0.8. With this bug, you can only put about half the number of models on the build plate at once because you now have to space them out further. I print all models at once and not one at a time, so no models should be high enough for the tool head to run into.

derbasteller, 2024-01-25:
Mein Slicer ist unter Windows 10 pro und kann zu Linux nichts sagen. Morgen werde ich auch mehrere Teile auf die Plattform legen. Mal sehen ob es bei mie auch Probleme gibt?

QIDITECH, 2024-01-26:
In 1.1.0, we added Z-axis lift optimizations, turning it off will solve your problem.
[image]

In addition, please provide your model and print parameters if possible so that we can optimize the program.

Vance1122, 2024-01-26:
It is the xmax3 and I am using the 0.6mm nozzle profile set for abs.

Vance1122, 2024-02-17:
This can be closed as it was fixed in version 1.1.1

closed 2024-02-17 by Vance1122