QIDITECH/QIDISlicer #30: Device list always empty v1.1.0

potens1, 2024-01-25:
I'm using the Linux appimage

My printer is setup and configured, but it does not appears in the `Device` tab.
[image]

[image]

Adding it manually in the `Device` tab, I can see the interface, but it is not saved and clicking another tab will loose it

[image]

[image]
The device is not present in the machine list

Clicking away and then on `Device` again brings me again to this screen
[image]

derbasteller, 2024-01-25:
Gib mal folgendes in deinem Browser ein: **http://192.168.3.229:10088**
Und prüf mal ob sich jedes mal die IP ändert?
Wenn dein Gerät im Router auf DHCP steht könnte nach einem Neustart eine andere IP vergeben werden.
Und dann geht die Verbindung verloren!

QIDITECH, 2024-01-26:
We can not reproduce your question, please tell me your Linux version and system architecture and other configurations.

rt1970, 2024-02-01:
Same Problem.
Update: i have used new name and not used DNS-name of printer. Only IP.
Now it show my printer

derbasteller, 2024-02-01:
DHCP ausgeschaltet? Feste IP? Direkt mit Kabel verbunden? WLAN ausgeschaltet?

QIDITECH, 2024-02-02:
@rt1970 Can you describe in detail how you triggered this bug, preferably with pictures.
thank you.

potens1, 2024-02-02:
> We can not reproduce your question, please tell me your Linux version and system architecture and other configurations.

I don't know if it was to me or to the German comment that I don't understand but here you are:

Arch Linux x64
Latest Qidi Slicer version (1.1.1) still has the issue.

I don't know what you need more picture I already put, in fact, it's what is already in your documentation.

rt1970, 2024-02-02:
@potens1 
I´m german too. "derbasteller" means you should control the network like DHCP, WIFI etc...
But we have another problem
I work with Windows 10 Pro x64 and QIDI slicer 1.1.1

QIDITECH, 2024-02-03:
rt1970 is right. But is your situation the same as his? In print_host, we added a judgment to exclude incorrect IP formatting, but we didn't take into account that you can also enter DNS-name, etc. Since the message pop-up in the photo you posted above just blocked the input box, I don't know if your question is consistent with rt1970. @potens1 
What are you inputing in Hostname, IP, URL?
[image]

potens1, 2024-02-04:
>  What are you inputing in Hostname, IP, URL? 
I'm using the hostname since the IP is assigned through DHCP and may, or may not get the same IP each time.
Using the IP (V4, and V4 only, ipv6 is not working either) does not show the bug, indeed. What is strange is, it works when using the hostname, until you switch tab

QIDITECH, 2024-02-07:
Ok, thank you for your cooperation, we will fix the problem in the next version.

whitekaren, 2024-03-21:
This problem has been solved in the 1.1.2 version of the software, you can download and use now, looking forward to your further feedback

closed 2024-03-21 by whitekaren