QIDITECH/QIDISlicer #31: machine list不能添加非IP設備

cavenlai, 2024-01-27:
當我使用mkspi或者mkspi.lan連接設備時，雖然能通過連接測試並顯示網頁，但是machine list中並沒有任何設備顯示。我使用的是動態IP，設備地址就要經常變更了。在1.07沒有machine list時一切是正常的。

QIDITECH, 2024-01-29:
请以压缩包的形式上传您的配置文件夹，以便我们复现并找出问题的原因。
[image]

cavenlai, 2024-01-29:
[QIDISlicer.zip](https://github.com/QIDITECH/QIDISlicer/files/14080775/QIDISlicer.zip)
[image]
这个并不会被记录到机械列表中，只要切换到其它页面就会丢失。

QIDITECH, 2024-01-31:
抱歉，我们暂时无法复现出您的问题，短时间内可能无法解决，请等待后续版本更新。

rt1970, 2024-02-01:
Same Problem! No Printer in List!

whitekaren, 2024-03-21:
This problem has been solved in the 1.1.2 version of the software, you can download and use now, looking forward to your further feedback

closed 2024-03-21 by whitekaren