QIDITECH/QIDISlicer #32: Crashing upon attempting to launch

blobabee, 2024-01-27:
v1.1.1 and v1.1.0 are both crashing when attempting to launch. I've attempted to delete the QIDI folder in the library/application support and other places to no success. I'm running on an M1 MacBook Pro 14" Sonoma 14.3

Here is the crash log:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               QIDISlicer [2141]
Path:                  /Applications/QIDISlicer.app/Contents/MacOS/QIDISlicer
Identifier:            nl.qidi.qidi
Version:               0.0 (0.0.0)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-01-27 12:23:07.1434 -0600
OS Version:            macOS 14.3 (23D56)
Report Version:        12
Anonymous UUID:        1F4264F3-C88B-3BD7-3110-3E7CD8D5D6F8

Sleep/Wake UUID:       2A151D1D-0683-4327-998C-7A2AF08B3B8E

Time Awake Since Boot: 3400 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /opt/homebrew/*/libzstd.1.dylib
Referenced from: <9328D2DB-E2DD-3E14-9BAD-286327D5B378> /Applications/QIDISlicer.app/Contents/MacOS/QIDISlicer
Reason: tried: '/opt/homebrew/*/libzstd.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/*/libzstd.1.dylib' (no such file), '/opt/homebrew/*/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file, not in dyld cache)
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	       0x18b9a255c __abort_with_payload + 8
1   dyld                          	       0x18b9aeb10 abort_with_payload_wrapper_internal + 104
2   dyld                          	       0x18b9aeb44 abort_with_payload + 16
3   dyld                          	       0x18b935584 dyld4::halt(char const*, dyld4::StructuredError const*) + 304
4   dyld                          	       0x18b932254 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 3884
5   dyld                          	       0x18b930edc start + 1844

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000006   x1: 0x0000000000000001   x2: 0x000000016afa2950   x3: 0x0000000000000076
    x4: 0x000000016afa2550   x5: 0x0000000000000000   x6: 0x0000000000000000   x7: 0x0000000000000c30
    x8: 0x0000000000000020   x9: 0x000000016afa24d1  x10: 0x000000000000000a  x11: 0x0000000000000000
   x12: 0x0000000000000032  x13: 0x1000000000000000  x14: 0x0000000000000004  x15: 0x0000000000008000
   x16: 0x0000000000000209  x17: 0x000000018b92e37c  x18: 0x0000000000000000  x19: 0x0000000000000000
   x20: 0x000000016afa2550  x21: 0x0000000000000076  x22: 0x000000016afa2950  x23: 0x0000000000000001
   x24: 0x0000000000000006  x25: 0x000000016afa3388  x26: 0x000000016afa32c8  x27: 0x0000000000000000
   x28: 0x000000001f070004   fp: 0x000000016afa2520   lr: 0x000000018b9aeb10
    sp: 0x000000016afa24e0   pc: 0x000000018b9a255c cpsr: 0x80001000
   far: 0x0000000000000000  esr: 0x56000080  Address size fault

Binary Images:
       0x104e5c000 -        0x107e23fff nl.qidi.qidi (0.0) <9328d2db-e2dd-3e14-9bad-286327d5b378> /Applications/QIDISlicer.app/Contents/MacOS/QIDISlicer
       0x18b92b000 -        0x18b9bf387 dyld (*) <50746901-db0e-39a0-b391-baaa6b82ad0f> /usr/lib/dyld
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
[... 136 lines cut ...]
      "treatmentId" : "52e9806b-48cf-4718-ba06-6aaa79468566",
      "experimentId" : "6553f184553f07431fda611f",
      "deploymentId" : 400000009
    }
  ]
}
}

Model: MacBookPro18,3, BootROM 10151.81.1, proc 8:6:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: ED242QR A, 1920 x 1080 (1080p FHD - Full High Definition), Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Sep  1 2023 19:46:07 version 20.10.1073.3.8.7.165 FWID 01-b66346c5
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB3.1 Hub
USB Device: easystore 266A
USB Device: USB2.1 Hub
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

pganns, 2024-01-27:
Crash in macos/Arm64/M2

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               QIDISlicer [20411]
Path:                  /Applications/QIDISlicer.app/Contents/MacOS/QIDISlicer
Identifier:            nl.qidi.qidi
Version:               0.0 (0.0.0)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-01-27 19:41:22.9380 -0300
OS Version:            macOS 14.2.1 (23C71)
Report Version:        12
Anonymous UUID:        21F2D4DB-BEDB-071B-7738-B1A21F45AE8A

Sleep/Wake UUID:       0DB3036A-44D4-45B2-B7DA-C0EE053B59D8

Time Awake Since Boot: 190000 seconds
Time Since Wake:       12781 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /opt/homebrew/*/libzstd.1.dylib
Referenced from: <0BE055C5-96C6-303F-B738-9788E2E7D3B0> /Applications/QIDISlicer.app/Contents/MacOS/QIDISlicer
Reason: tried: '/opt/homebrew/*/libzstd.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/*/libzstd.1.dylib' (no such file), '/opt/homebrew/*/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file, not in dyld cache)
(terminated at launch; ignore backtrace)

blobabee, 2024-01-28:
Not sure what to make of that, but just to confirm, it truly is an M1 chip. I used to work for Apple service (not that you need that to know lol)

mrunion, 2024-01-28:
That was happening for my -- it's missing a library. Install zest and it will work.

If you are using Homebrew, just run:

`brew install zstd`

That should fix it.

blobabee, 2024-01-28:
Hi. I am not. Not even sure what that is. lolSent from my iPhone

mrunion, 2024-01-28:
No worries. It's just software that has some popular things pre-packaged for Mac. You can read about it here if you're interested:

https://brew.sh/

At any rate, installing the zstd library should fix this issue.

If you prefer not to use home-brew, you can go to this site and should be able to get the Zstd library from there:

https://facebook.github.io/zstd/

blobabee, 2024-01-29:
Ok. I can def do that. Though shouldn’t the app be self-contained like all prior releases or are you just suggesting this as a work-around till it’s fixed?

Thanks much!

https://brew.sh/

https://facebook.github.io/zstd/

QIDITECH, 2024-01-29:
Which architecture of software do you install? We provide x86 and arm64.

blobabee, 2024-01-29:
As always, I chose ARM for the M1Sent from my iPhone

ReikiDave, 2024-01-29:
Mine is crashing also.
Yesterday version 1.0.8 crashed after a window opened telling me there was a new version available.  I couldn't

ReikiDave, 2024-01-29:
Windows event log:

Log Name:      Application
Source:        Application Hang
Date:          1/29/2024 7:15:19 AM
Event ID:      1002
Task Category: (101)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      HP-Home-Workstation
Description:
The program qidi-slicer.exe version 1.1.0.0 stopped interacting with Windows and was closed. To see if more information about the problem is available, check the problem history in the Security and Maintenance control panel.
 Process ID: 4094
 Start Time: 01da52acc2028dd6
 Termination Time: 4294967295
 Application Path: C:\Program Files\QIDISlicer\qidi-slicer.exe
 Report Id: 7e18c95e-ccc1-4dc6-960a-f3a5ab8cb8e1
 Faulting package full name: 
 Faulting package-relative application ID: 
 Hang type: Top level window is idle

Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Hang" />
    <EventID Qualifiers="0">1002</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>101</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2024-01-29T12:15:19.0809638Z" />
    <EventRecordID>651556</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>HP-Home-Workstation</Computer>
    <Security />
  </System>
  <EventData>
    <Data>qidi-slicer.exe</Data>
    <Data>1.1.0.0</Data>
    <Data>4094</Data>
    <Data>01da52acc2028dd6</Data>
    <Data>4294967295</Data>
    <Data>C:\Program Files\QIDISlicer\qidi-slicer.exe</Data>
    <Data>7e18c95e-ccc1-4dc6-960a-f3a5ab8cb8e1</Data>
    <Data>
    </Data>
    <Data>
    </Data>
    <Data>Top level window is idle</Data>
    <Binary>54006F00700020006C006500760065006C002000770069006E0064006F0077002000690073002000690064006C00650000000000</Binary>
  </EventData>
</Event>

ReikiDave, 2024-01-29:
Problem fixed after uninstalling and deleting all files, clearing out the recycle bin.
Restarted the computer.
Reinstalled version 1.1.1
Restarted the computer again.

All seems to be working now.  Don't know what the hang-up was.

QIDITECH, 2024-01-31:
@ReikiDave Your situation is different from the above, only the files are missing when you install.

QIDITECH, 2024-01-31:
@blobabee Please try to see if prusa slicer works properly on your computer.
We can not reproduce your problem, it may take some time to resolve this issue, please wait for a later version upgrade.

blobabee, 2024-01-31:
Hi,I have Prusa Slicer and it’s up to date and works. Just like the previous version of QIDi Slicer did. I was watching for a newer version of QIDi Slicer and firmware as I normally do, so when this latest version was available, I installed it like all the past versions and this is the first to exhibit this problem. Best,BobSent from my iPhone

ReikiDave, 2024-01-31:
I'm not sure what was causing the issue.
Everything worked fine once I uninstalled, restarted my computer, reinstalled, and restarted my computer.

Thank you all for checking in

Tkmacny, 2024-02-05:
QIDISlicer_1.1.1_MacOS_arm64 crashes upon launch but QIDISlicer_1.1.1_MacOS_x86 works so far. Have only sliced model.

MacBook Pro 14 with M2 Max
Sonoma 14.1.1

blobabee, 2024-02-06:
Same here! (as I believe I mentioned prior). Awaiting an Arm64 update

QIDITECH, 2024-02-07:
There seems to be something missing in the dependencies of the new version that is causing compatibility issues with some macos. We are working to resolve this issue, so please use v1.0.8 until then.

mrunion, 2024-02-14:
I am still having issues with PLA. I have found that getting PLA to stick in the middle of the print bed is fine — that’s why my little test cubes print OK.

The problem comes in when I try to print large items. It seems that the front left of the print bed is having issues heating like the rest of the bed.

I have attached a picture of a larger, 150mm x 150mm single layer print and you can see that in the bottom left side of the photo (that corresponds to the 0,0 corner of the printer) the PLA does not adhere well, but the right and back sides do. It seems like the heated bed may be defective?

This was printed with the default filament settings for generic PLA (pictured as well).

I will admit I am not a “professional” — but this is my 6th 3D printer. I feel that my Ender 5 Plus is much less finicky than this printer. Am I just missing something with how I am trying to rectify this problem?

￼

￼

Thanx,
Matt Runion
***@***.***

mrunion, 2024-02-14:
As a followup, you can see how wildly out of level the print bed is in this tuning image. The photo of the print in the previous email got rotated 90 degrees, so I have reposted the print here. As you can see, the “bad print” areas line up almost exactly with the high corner in the tune image. (I have circled the point on the print that corresponds to the front left corner of the tune image.) The bed is off over 0.23 mm — which is more than the layer height of a default layer.

Is there a way to verify the bed mesh is compensating for the unevenness of the bed?

I think my print bed is defective.

￼

￼

Thanx,
Matt Runion
***@***.***

QIDITECH, 2024-02-14:
@mrunion  Sorry, I can't see the picture you uploaded, and this is another issue unrelated to this issue, please ask the seller to adjust your printer to make it work better.
https://qidi3d.com/pages/warranty-policy-after-sales-support

mrunion, 2024-02-14:
Apologies — I used the wrong email.

I have sent it to the correct folks. And my post can be deleted. Sorry for the troubles.

Thanx,
Matt Runion
***@***.***

whitekaren, 2024-03-21:
In the latest 1.1.2 macOS arm64 version of the software has removed the dependency on the zstd library file, you can now download and use, looking forward to your further feedback

closed 2024-03-21 by whitekaren

blobabee, 2024-03-21:
Installed and started up properly! Will provide more feedback this weekend when I get a chance to use itThank you!BobSent from my iPhone