QIDITECH/QIDISlicer #36: Printer Upload not possible, if system proxy set

Quantomax, 2024-02-02:
Config:
======
QIDIslicer on IP 192.168.1.5 (Win64, V1,0.7 and V1.1.1)
X-Smart3 on IP 192.168.1.248

Environment Vars:
HTTPS_PROXY=http://proxy.localnet:3128/
HTTP_PROXY=http://proxy.localnet:3128/
NO_PROXY=localhost,127.0.0.1,192.168.1.0/24,192.168.2.0/24

Windows Settings (no proxy for local addresses as above)
proxy.localnet resolves to 192.168.3.2, can't reach 192.168.1.x normally.

Situation:
======
Sending data to printer over LAN will fail, as proxy is used regardless of no_proxy settings.
If I change firewall rules, so the then proxy can reach 192.168.1.248:10088 printer "Test" will work and finds printer.
If I upload data to printer data will be send to proxy-ip instead of to printer via proxy and fails (http error 503)

QIDITECH, 2024-02-03:
Changing the rules for sending data is difficult, but you can upload and print files with fluidd.

Quantomax, 2024-02-03:
The problem seems to be related to some lib, that reads environment-vars case sensitive and matches only exact IPs.

Solution is a lowercase "no_proxy" environment var which contains the (fixed) ip of the printer, in my case "192.168.1.248". DNS-Names and subnets will not work.

closed 2024-02-03 by Quantomax