QIDITECH/QIDISlicer #38: QIDI ASA profile chamber heat problem!

JohnPetrucci, 2024-02-17:
With the default Qidi ASA profile and the chamber heating set to 55 degrees, once the print is started, the chamber heating does not activate automatically."

derbasteller, 2024-02-17:
Hallo,

 

die erste Schicht wird immer ohne Kammerheizung gedruckt!

Ab der 2. Schicht muss die Kammerheizung hochfahren.

 

thank you and stay healthy!,

danke und bleiben sie gesund! 

Michael Kisser

Tel.:   0 51 45 / 93 94 0 - 91

 

Mobil: 0 1575 11 15 545

Mail:  ***@***.*** ***@***.***> 

 

Auf dem Stüh 21

29353 Ahnsbeck

 

Diese E-Mail enthält vertrauliche und/ oder rechtlich geschützte Informationen. Wenn Sie nicht der richtige Adressat sind oder diese E-Mail irrtümlich erhalten haben, informieren Sie bitte sofort den Absender und vernichten Sie diese Mail. Das unerlaubte Kopieren sowie die unbefugte Weitergabe dieser e-Mail ist nicht gestattet.

This e-mail may contain confidential  and/ or privileged information. If you are not the intended recipient (or have received this e-mail in error) please notify the sender immediately and destroy this e-mail. Any unauthorized copying, disclosure or distribution of the material in this e-mail is strictly forbidden.

QIDITECH, 2024-02-19:
The chamber will activate when the second layer is printed.

closed 2024-02-22 by QIDITECH