QIDITECH/QIDISlicer #40: QIDISlicer 1.1.1 pressure advance calibration issue

Vance1122, 2024-02-17:
When doing the pressure advance calibration, on both the lines and pattern option, the column of numbers is printer first and they are about 4mm tall. Then the tool head comes back to print the lines or pattern depending on which one you picked and the tool head runs into the numbers. In my case, the column of numbers stuck well enough for the tool head to move the build plate over on the platform. Please fix this. The number column could be 1mm tall from the surface or shorter and still be readable.

QIDITECH, 2024-02-19:
Thanks for your feedback, we will fix the issue in the next release.

whitekaren, 2024-03-21:
The 1.1.2 version of the software has fixed the problem you described, you can download it and use it now, looking forward to your further feedback

closed 2024-03-21 by whitekaren