QIDITECH/QIDISlicer #41: QIDI Slicer 1.1.1  Max Volumetric Speed calibration

Vance1122, 2024-02-17:
This test really needs to be run in vase mode. Having it stop on one layer and then start the next layer in a random place is broken and leaves a lot of questions about where the MVS maxes out.

Vance1122, 2024-02-17:
OH. And if you try to change the print settings to enable vase mode, it gives an error that vase mode only works with single material printing even though we are using one filament to do our calibrations.

QIDITECH, 2024-02-19:
I'm sorry, I don't understand.
Our calibration procedure did not activate vase mode.

Vance1122, 2024-02-19:
But it should. This calibration is supposed to be in vase mode.

QIDITECH, 2024-02-19:
As you know, vase mode may be more suitable for printing, but it is not easy to adjust the speed.

closed 2024-02-19 by QIDITECH