QIDITECH/QIDISlicer #43: QIDI SLicer-1.1.1 Loss of part picture on X-plus 3

rayzor120, 2024-02-20:
Ever since the update, the part not showing in the display window. Old files before the update can still display part.

X-plus 3 running 
SOC: V4.2.13
MCU: V0.10.0
UI: V4.2.13

[image]

cl4nspi, 2024-02-20:
https://github.com/QIDITECH/QIDISlicer/issues/35#issuecomment-1919652036

rayzor120, 2024-02-20:
I agree.

rayzor120, 2024-02-20:
Still trying to find the location in printer settings.

cl4nspi, 2024-02-20:
> Still trying to find the location in printer settings.

Printer Settings -> general 
The setting is only visible in expert mode

QIDITECH, 2024-02-20:
Update the slicing software configuration.
In order to better fit fluidd, the parameter format of the v1.1.1 preview has been changed. After the software is updated, the parameter configuration also needs to be updated for normal use.

closed 2024-02-20 by QIDITECH