QIDITECH/QIDISlicer #44: Please add a "shrinkage" value for the filament types

Phil1988, 2024-02-21:
In SuperSlicer and OrcaSlicer you have a shrinkage option in the filament section.

This is important if you want accurate objects from materials that shrinks while cooling (like ABS/ASA, etc)
It would be nice to have this feature availible in QIDISlicer as well.

Here are examples of SuperSlicer at ABS:
[image]

Or from OrcaSlicer at ABS:
[image]

QIDITECH, 2024-02-22:
Ok, we'll add and test this parameter.

whitekaren, 2024-03-21:
You can now set the scale of the model according to the shrinkage rate of the filament in version 1.1.2 of the software for more accurate printing (thanks to OrcaSlicer), looking forward to your further feedback

closed 2024-03-21 by whitekaren

Phil1988, 2024-03-21:
@whitekaren You meant "thanks to SuperSlicer/supermerill" right?
Credits to the ones that have done it ;)

https://github.com/supermerill/SuperSlicer/issues/287
(Orcas parameter name is even the same as from SuperSlicer: "filament_shrink")