QIDITECH/QIDISlicer #46: Bug between "Perimeter Generator Arachne" and "Only one wall on top surfaces"

BlackThumb, 2024-02-28:
Example:
If I set 4 external perimeters with "Only one wall on top surfaces = all top" and "Perimeter Generator = Arachne" all works fine and in the slice mode you can see all the 4 perimeters and all the top surfaces with 1 perimeter, if I switch on "Perimeter Generator = Classic" all the model has only 1 perimeter and not only the Top.
Then if I keep "Perimeter Generator = Classic" and switch "Only one wall on top surfaces = Only Top Most" the model has again 4 perimeters and only the top most surface has 1 perimeter only.

QIDITECH, 2024-02-29:
Thank you for your feedback, we have fixed this bug, please wait for the next version update.

whitekaren, 2024-03-21:
The 1.1.2 version of the software has fixed the problem you described, you can download it and use it now, looking forward to your further feedback

closed 2024-03-21 by whitekaren