QIDITECH/QIDISlicer #47: CPU usage and more

bathrobehero, 2024-02-28:
QIDISlicer idling pegs a single CPU thread after doing the following:

 - open QIDISlicer (no need to open any project), cpu usage is 0
 - click on the Device tab
 - click on any other tab and CPU usage goes high and stays there.

Weird thing is if you click on the Device tab again, CPU usage goes down to 0, until you click on any other tab again. If you never click on the Device tab, the issue never occurs.

QIDITECH, 2024-02-29:
To ensure the stability of the fluidd connection, the thread is always running in the background after startup. If a physical printer is not connected, the thread will not start.

melrhombus, 2024-03-26:
> To ensure the stability of the fluidd connection, the thread is always running in the background after startup. If a physical printer is not connected, the thread will not start.

If I use PrusaSlicer and point to a Qidi machine via moonraker there is no CPU spike nor is there one when I load fluidd via an external browser. 

If QidiSlicer is idle my desktop starts increasing the fan speeds and I see a constant 15-20% CPU usage with it just running in the background. This wasn't the case several versions ago though... something in the last 2 or so versions this started occurring.

Maxwell-Wellan, 2024-06-03:
I am encountering the same issue. Its at the point where my other processes will lag due to the overuse of the Qidi Slicer. The only viable option is to close the software when not slicing or checking on the status of a print which is not ideal.

QIDITECH, 2024-07-08:
We have fixed this bug, please update the software to the latest version.

closed 2024-07-08 by QIDITECH

QIDITECH, 2024-07-09:
OK, your problem is different from everyone else's, we reference a new issue and keep trying to solve it.