QIDITECH/QIDISlicer #51: New release 1.2 appimage doesn't work on Manjaro Linux 23.1.3

Kenrgoss, 2024-03-21:
Just the title really. I was excited to see this release as it said it had some fixes for Linux. I've been using 1.0 since 1.1 would frequently crash and was hopeful that this new release wouldn't have that issue. It doesn't launch at all for me though. Please let me know if there's any commands or debugging info I can share to assist. Thanks!

QIDITECH, 2024-03-25:
We tried to add libwebkit2gtk-4.0 to the installation package, but this seemed to cause a bigger problem.
We have re-uploaded the installation package, please re-download and run the software.
Both of your issues should be fixed in this new version, if there are still problems, please tell us more details.

Kenrgoss, 2024-03-28:
I downloaded it and tried to run it again, but it still fails to launch. What additional details would be useful for you to know?

QIDITECH, 2024-03-29:
Did you download the software from github? There is a delay in updating the installation package on the official website.
If you have downloaded from github then your problem seems to be different from others, please run the software through terminal and attach a screenshot to let me know what dependencies are missing.

closed 2024-11-20 by QIDITECH