QIDITECH/QIDISlicer #52: QIDISlicer_1.1.2 AppImage fails to open on Arch Linux

davtechsupport, 2024-03-21:
get this error: ./QIDISlicer_1.1.2_Linux_5fbe3031dd02701756bdef392e403007.AppImage /tmp/.mount_QIDISlRj7lA8/usr/bin/bin/qidi-slicer: error while loading shared libraries: libicuuc.so.66: cannot open shared object file: No such file or directory

I've confirmed icu and lib32-icu are installed.

grimpirate, 2024-03-22:
I use Arch Linux as well and can confirm I received the same error. I've also confirmed installation of icu and lib32-icu, furthermore version 1.1.1 is working.

QIDITECH, 2024-03-23:
We tried to add libwebkit2gtk-4.0 to the installation package, but this seemed to cause a bigger problem.
We have re-uploaded the installation package, please re-download and run the software.

closed 2024-03-25 by QIDITECH