QIDITECH/QIDISlicer #53: Prusa Slicer License attributions were incorrectly removed

stew675, 2024-03-21:
In this PR change set here: https://github.com/QIDITECH/QIDISlicer/commit/cad0b137b44e0073d8b79751b83027060ad5716a

It looks like someone did a blanket find/replace of Prusa for Qidi that looks like it was intended to catch variable names in the source code, but applied it so thoroughly that it ended up editing out the Open Source Licensing Attribution in the About file here: https://github.com/QIDITECH/QIDISlicer/blame/516d3a3313f4fde306f918f7a36207bf5f98d42b/src/slic3r/GUI/AboutDialog.cpp#L279

Items like this should be corrected as it is causing negative discussion in the community that Qidi is not correctly PrusaSlicer in the licensing attributions.

QIDITECH, 2024-03-22:
I'm very sorry, this is our mistake, we have modified and re-uploaded the code. The released version 1.1.2 will not be modified, and re-uploading files may cause some customers to be unable to update online.
The problem will not be closed, it will remind us not to make the same mistake again.

stew675, 2024-03-22:
Thank you for the reply and fast fix in good faith.

You may also want to look over that pr https://github.com/QIDITECH/QIDISlicer/commit/cad0b137b44e0073d8b79751b83027060ad5716a in more detail.  It also looks like a number of web links to the Prusa Knowledge database got scribbled over, and since Qidi does not have an equivalent version of that database, most of those Prusa Knowledge Database links now point to nowhere, and are consequently broken.

QIDITECH, 2024-03-23:
Yes, we will gradually improve everything, but it will take some time.