QIDITECH/QIDISlicer #54: OS X Appearance not changing

wabbitguy, 2024-03-21:
Mac OX - Sonoma - X86
```
Version: 1.1.2
Build: QIDISlicer-1.1.2+Win64
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Version 14.3.1 (Build 23D60)
Total RAM size [MB]: 17,180MB

OpenGL installation
GL version: 4.1.0
Profile: Core
Vendor: ATI Technologies Inc.
Renderer: AMD Radeon Pro 5300 OpenGL Engine
GLSL version: 4.10.0
Textures compression: Enabled
```

Changing the OS X System Preferences - Light or Dark only affects the object/preview. All other settings are unaffected. Makes it very difficult to read some of the options text that appear in green (previous QIDI slicer displayed properly with "Dark" theme).

[image]

Extremely difficult to read:

[image]

majo1604, 2024-03-22:
Same issue here 😑

QIDITECH, 2024-03-23:
We are very sorry, but in order to be compatible with the system version 10.13-11, we reduced the version of the SDK package and some dependencies, which caused this issue. We will solve this problem as soon as possible.

QIDITECH, 2024-03-28:
We re-uploaded the installation package, please re-download and install the software.

closed 2024-03-28 by QIDITECH

wabbitguy, 2024-03-28:
@QIDITECH downloaded and it's working perfectly. Thanks!