QIDITECH/QIDISlicer #65: QidiSlicer is launching, but main window not displayed

tohoku, 2024-05-28:
QidiSlicer: 1.1.4 for MacOS (Arm64)
Operating System: MacOS Sonoma 14.5
Computer: Mac Studio M1

**Steps to reproduce:**
1. Start QidiSlicer from _Applications_
2. (Splash screen is shown.)
3. Launch completes.
4. Confirming the menu bar, "QidiSlicer" as a menu item is displayed to the right of the Apple logo. The other menu items are also available. The main window (with the build plate) is not visible.

I also have an old Intel Mac, but the problem does not occur on that computer.

tohoku, 2024-05-28:
I found the window, it was about 2 pixels wide.

Feature request: Add a way to reset the UI (restore window sizes).

QIDITECH, 2024-05-30:
Please try using the x86 version of the software.
Please also attach a screenshot of the window and we will try to resolve the issue.

tohoku, 2024-05-31:
[image]

I didn't take a screenshot when the problem happened, but I created this illustration instead. The main window was just a narrow line on the screen (2-3 pixels wide), so I didn't even see it.

waters0, 2024-09-04:
Just happened to me.

QidiSlicer: 1.1.5 for MacOS (Arm64)
Operating System: MacOS Sonoma 14.6.1
Computer: MacBook Pro 16-inch 2021

If it happens again I'll capture a screenshot.