QIDITECH/QIDISlicer #67: Qidislicer 1.1.4 menu item not showing / weird

GXMV1978, 2024-06-03:
Near the calibration tab that's non fonctionning for me (issue #66 ), there is an hidden tab that you can see at mouse over. If you click it, one set of brackets appear, if you click again, 2 sets of brackets, click again and 3 sets, and so on and so forth.
[image]

QIDITECH, 2024-06-04:
Please close ‘Set settings tabs as menu items’.
The calibration does not yet support this setting.
[image]

closed 2024-06-04 by QIDITECH

GXMV1978, 2024-06-04:
Thanks a lot have a great day