QIDITECH/QIDISlicer #69: 3mf files not opening from file explorer

smitdylan2001, 2024-06-15:
Ever since the 1.1.4 update 3mf folders on my drive do not open. When selecting open with - Qidi slicer it does not do anything either

QIDITECH, 2024-06-18:
Which version did you update to 1.1.4?
And can you provide your 3mf file?

smitdylan2001, 2024-06-18:
I used 1.1.3 before (first time using qidislicer)
For me any 3mf file made from saving an opened stl does not open by double clicking. I have to open the slicer and open it via the recent tab
[Mask.zip](https://github.com/user-attachments/files/15883238/Mask.zip)

QIDITECH, 2024-06-20:
We can open these files normally. According to your description, I think there is a problem with the registry when you install the software. Please try to uninstall the software, then restart the computer and reinstall the software.

smitdylan2001, 2024-06-24:
> We can open these files normally. According to your description, I think there is a problem with the registry when you install the software. Please try to uninstall the software, then restart the computer and reinstall the software.

Still does not work. Could it be that it is because I use D://Programs/QIDISlicer as path?

QIDITECH, 2024-06-25:
The installation path does not affect this.
Could you make a video showing me the problem? Maybe there are some details you didn't describe clearly.

smitdylan2001, 2024-07-16:
Here you go.
3mf files made in 1.12 or 1.14 do not open in 1.15 for me. Projects made in 1.15 work fine now
https://github.com/user-attachments/assets/57d25dbe-e502-4453-a8c0-2387b221e0da

QIDITECH, 2024-07-17:
Have you tried to re-associate files?
[image]

And can you open 3mf by specifying the application in Open With?
[image]

smitdylan2001, 2024-07-17:
Yes, and I even reinstalled the slicer. It works with 1.15 3mf files, but
not older ones

QIDITECH, 2024-07-19:
We have understood your situation, but we are still unable to repeat the problem. Your problem is not a normal logical problem, which is very strange and seems to happen only on your computer. We will continue to explore the issue and will contact you once we find a solution.