QIDITECH/QIDISlicer #70: Qidi Slicer Crashing

beatcl9, 2024-06-17:
Hi there, new-ish user here. I have been  using QIDI slicer on my new X-Plus 3 for the past 2 weeks. Today when I try to open Qidi Slicer, the app launches but the curser is stuck with a loading icon. Closing and researting the application does not solve this. I have also uninstalled and reinstalled the application which did not work. The cursor is stuck with loading icon and I can not use the application. Please advise, thank you! 

My slicer is the newesr version and was working perfectly fine yesterday.

QIDITECH, 2024-06-18:
Could you please provide a video or picture to more clearly describe your problem?
And tell me your computer system version and configuration.
Also, did anything unusual happen the last time you used it?

beatcl9, 2024-06-18:
Thank you for your response. Nothing unusual happened the last time that I
used it. I had saved my last project and closed the program normally.
Please see the video attached.

QIDITECH, 2024-06-20:
Github cannot view videos attached to emails, please use another way to share your videos.

beatcl9, 2024-06-21:
Hi I have uploaded a new video to my google drive folder linked here.

https://drive.google.com/drive/folders/1--q0bxvZ86edeZx462WTusrt1uAfL0uf

Please let me know if you can not access it. I have set the permission so
that anyone with the link can view the folder.

Thanks

QIDITECH, 2024-06-21:
[image]
1. Please open the path in the picture and send the configuration file to me.
2. After backing up the configuration file, delete all files in the directory and start the software again.

beatcl9, 2024-06-22:
This is what I could find. Is this what you were looking for?
 api-ms-win-crt-runtime-l1-1-0.dll
<https://drive.google.com/file/d/1yOTCF5f7jNJNku2aPaxotybGmLWtX1cR/view?usp=drive_web>
 libgmp-10.dll
<https://drive.google.com/file/d/13tPLafIMuZgrqdTYmJr6ndSp7vZrzvc2/view?usp=drive_web>
 libmpfr-4.dll
<https://drive.google.com/file/d/1085W4f5NKPh8ThvxUzMXpWrkE62sjORe/view?usp=drive_web>
 msvcp140.dll
<https://drive.google.com/file/d/1rGJhvviLBsLcFXLkvWvaTuEWkeBs7d5P/view?usp=drive_web>
 msvcp140_codecvt_ids.dll
<https://drive.google.com/file/d/1uRraXgjXbzlx4lRPeDWtezA_txX0HzOO/view?usp=drive_web>
 OCCTWrapper.dll
<https://drive.google.com/file/d/15Yq83s32-UA83EHtI5v4sT-qSsswIjQu/view?usp=drive_web>
 qidi-gcodeviewer.exe
<https://drive.google.com/file/d/1F9CKuFsfhDuRSijHesuuhGHkuu6LMR1l/view?usp=drive_web>
 qidi-slicer.exe
<https://drive.google.com/file/d/1CUmTSM_ctyXc4MmHEAyLQuGjXg-8OBq-/view?usp=drive_web>
 QIDISlicer.dll
<https://drive.google.com/file/d/1FBdVPu-I9_nOqmF6Ozg6Al2xjWzkeUZk/view?usp=drive_web>
 QIDISlicer.url
<https://drive.google.com/file/d/1iMEkVE6Xi5wZWvCsqYbqQS4lGa71lrl1/view?usp=drive_web>
 uninst.exe
<https://drive.google.com/file/d/1CCfqyi1Ixv6ZHW0Mc9GyNVhCJ1MQA0FI/view?usp=drive_web>
 vcruntime140.dll
<https://drive.google.com/file/d/1aCdbAk_WnLALY9nuP4w_XdOhbLLbNnKp/view?usp=drive_web>
 vcruntime140_1.dll
<https://drive.google.com/file/d/1UIMGOUVM1Fgsnk-hYqZER23C7_kmg2Cm/view?usp=drive_web>
 WebView2Loader.dll
<https://drive.google.com/file/d/17vvesAV7-7tdx2MdU4y6GTl0S6-6ok70/view?usp=drive_web>

QIDITECH, 2024-06-24:
These are the files in the software installation directory.
Configuration files are automatically generated when the software is running. They are saved in a fixed place. Please follow the path in the picture above to find them.

beatcl9, 2024-06-24:
[image: image.png]
What is the specific path and what folder are the configuration files in?
Which folder is the one I need to open? I sent you everything I had in my
qidislicer folder.

QIDITECH, 2024-06-24:
C:\Users\Administrator\AppData\Roaming\QIDISlicer
Administrator is your computer user name.
AppData folder may be a hidden folder, and you need to display hidden files to access it.

sktpapas, 2024-07-02:
It is not clear if it is the same issue, but after updating Windows 11 June 25, 2024-KB5039302, the QIDI slicer now crashes.

QIDITECH, 2024-11-20:
No follow-up response, close the issue.

closed 2024-11-20 by QIDITECH