QIDITECH/QIDISlicer #72: ERROR voronoi cell does not have a source point

STEVECORNETTE, 2024-06-22:
Haven't received my X Max 3 yet. I am a Prusa Slicer veteran of 3 years. Thought I would install Qidi Slicer and get familiarized.. AND this error comes up: 

ERROR Voronoi cell does not have a source point!. 

How do I fix this? When the printer arrives early next week I need to be printing parts!
Please help.

Steve C.

QIDITECH, 2024-06-24:
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md
Please refer to prusa's build documentation.

QIDITECH, 2024-06-24:
Did the problem occur while you were running the software, or during use? Can you provide more detailed videos or pictures to describe this problem? I need to know how this problem happened, or how you caused it.

STEVECORNETTE, 2024-06-24:
While I am running software. It will slice fine with 4 instances but when I
add a 5th and reslice the error occurs.

[image: image.png]

STEVECORNETTE, 2024-06-24:
AND.. I've just gotten it to create the error when I rotate the instances..
I cannot have these errors when this printer arrives
tomorrow. I need to be printing parts. :-(

QIDITECH, 2024-06-25:
I am very sorry that I cannot view the picture you sent, please resend it.
Can you provide your model or 3mf file so that we can reproduce the problem more accurately and solve it?
And you can try to use the copy and paste function, maybe you can avoid this bug for a while.
[image]

STEVECORNETTE, 2024-06-25:
Thank you for your help! 
Sometimes the error comes up when I add or delete instances, rotate them or manipulate them in some way.
My guess is that it is related to generating infill patterns. I will attempt to send the 3mf file.
[image]
[voronoi-error.docx](https://github.com/user-attachments/files/15970812/voronoi-error.docx)

STEVECORNETTE, 2024-10-28:
> Hi, I have the same problem. Attached you find the model. Can you please
> tell me, how to solve the problem
> Polygonoid_Lampe_V2.zip

Ok, I'll take a look.

QIDITECH, 2024-10-29:
The problem still cannot be repeated, we will try to fix it in the next version, but we need your help to test it.

STEVECORNETTE, 2024-10-29:
Ok, let me know.

QIDITECH, 2024-11-20:
We fixed this issue in the latest version, but we are not sure if it will work, please download the latest version.
https://github.com/QIDITECH/QIDISlicer/releases
If there are no problems, I will close the issue after a week.