QIDITECH/QIDISlicer #73: Klippy not connected

C3362, 2024-06-22:
Hi
I have QIDI Smart 3 printer and have problem running QIDI Slicer on Ubuntu 22.04. Odd on my old Ubuntu 20.04 everything was OK.

After Printer setup I see 
[image]

A was able print but each time I want to check "Device" tab it shows that error or does not show anything at all.
Is something missing on system?

Logs from command line:
`➜  ~ QIDISlicer_1.1.4_Linux.AppImage

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:02.899: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:02.899: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:02.900: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:02.900: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:02.904: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:02.904: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:02.905: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:02.905: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:02.909: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:02.909: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:02.910: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:02.910: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
No elefant_foot_compensation_layers in ConfigOptionsGroup config.
No precise_z_height in ConfigOptionsGroup config.

** (qidi-slicer:12584): CRITICAL **: 13:29:03.220: Cannot register URI scheme wxfs more than once

** (qidi-slicer:12584): CRITICAL **: 13:29:03.220: Cannot register URI scheme memory more than once

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:03.298: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:03.298: gtk_widget_set_size_request: assertion 'width >= -1' failed

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:03.314: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:03.314: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:03.314: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:03.314: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:03.316: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:03.316: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:03.317: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:03.317: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:03.318: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:03.318: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:03.318: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:03.318: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:03.318: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:03.318: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:12584): GLib-GObject-WARNING **: 13:29:03.318: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:12584): Gtk-CRITICAL **: 13:29:03.318: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
01:29:05 PM: Debug: window wxTreeCtrl@0x556036d0e1d0 ("treeCtrl") lost focus even though it didn't have it
01:29:05 PM: Debug: window wxTreeCtrl@0x556036d0e1d0 ("treeCtrl") lost focus even though it didn't have it
01:29:05 PM: Debug: window wxTreeCtrl@0x556036d0e1d0 ("treeCtrl") lost focus even though it didn't have it
01:29:05 PM: Debug: window wxTreeCtrl@0x556036d0e1d0 ("treeCtrl") lost focus even though it didn't have it
01:29:05 PM: Debug: window wxTreeCtrl@0x556036d0e1d0 ("treeCtrl") lost focus even though it didn't have it
01:29:05 PM: Debug: window wxTreeCtrl@0x556036d0e1d0 ("treeCtrl") lost focus even though it didn't have it
01:29:05 PM: Debug: window wxTreeCtrl@0x556036d0e1d0 ("treeCtrl") lost focus even though it didn't have it
`

QIDITECH, 2024-06-24:
We are aware of this issue, but there are too many versions of the linux system, and different versions have different issues, we will resolve this issue as soon as possible, until then please connect through a browser and visit fluidd.

QIDITECH, 2024-11-20:
We have uploaded two Linux versions compatible with libwebkit2gtk-4.0 and libwebkit2gtk-4.1 in the latest test version, please download and test.
https://github.com/QIDITECH/QIDISlicer/releases
If there are no problems, I will close the issue after a week.