QIDITECH/QIDISlicer #76: Magikarp body repaired incorrectly

carol8, 2024-06-23:
Hi,

I tried to slice a broken STL and it repaired it wrongly, despite saying it repaired some edges:
[image]
[image]

When loading this file in OrcaSlicer, repairing it there and slicing it using QIDISlicer I'm getting the correct result:
[image]
[image]

The broken STL and the correct STL are available [image]
If you have any questions about reproducing the error, ask and I'll respond ASAP. Thanks a lot!

QIDITECH, 2024-06-24:
Can you provide this model to us? If you do not want to disclose this model, you can send it to our after-sales mailbox.

carol8, 2024-06-24:
The model is available in the original message (see the underlined, blue link with the text "here" at the bottom of the message). If you are not able to click on the link, copy and paste it from here: [https://drive.google.com/drive/folders/1QYCK-e3OROqIqV4NlY9NKu7OtprteO0-?usp=sharing](https://drive.google.com/drive/folders/1QYCK-e3OROqIqV4NlY9NKu7OtprteO0-?usp=sharing)

QIDITECH, 2024-06-25:
We were unable to reproduce the problem you said, the fix function on my computer can work properly and get correct results with Orca.

QIDITECH, 2024-06-25:
Please try again to fix the model, and if that doesn't work, please record a video or screenshot to show me where the error occurred.

carol8, 2024-06-25:
The fix function in Orcaslicer is working properly, but the fix function in QIDISlicer does not. I will record a video and send it to you later today. Thanks a lot!

closed 2024-09-30 by QIDITECH