QIDITECH/QIDISlicer #77: QIDI slicer 1.1.4 - Error 500 when uploading to printer

TinkerBarn, 2024-06-28:
When uploading a sliced project file via LAN/WLAN, error 500 occurred with a long error code. It was difficult for me as a user to recognize the reason for this error.

After a long search, I found the reason. There was already a print file with an identical name '3DBenchy.gcode' on the Q1 Pro printer, I changed the file name in the upload pop-up window to 'My3DBenchy.gcode' and there was no more error message.
The reason may be that the print files supplied in the firmware are read-only.

Request: Please implement an error message "File name already exists" or "This file name must not be used" in the next version.

QIDITECH, 2024-06-29:
This is not a software problem, this is an error message returned on the firmware.
The issue has been fixed, please try updating the firmware to the latest version.

TinkerBarn, 2024-06-29:
My QIDI Q1 Pro is already updated to the latest available firmware V4.4.20
So it seems it is not fully fixed in the latest firmware.

QIDITECH, 2024-07-01:
Thanks for your feedback, we will review the issue and fix it in the next release.

closed 2024-09-30 by QIDITECH