QIDITECH/QIDISlicer #78: appimage won't run, looking for shared library libwebkit2gtk-4.0.so.37

ChrisJones79, 2024-06-30:
This library didn't come with my standard Debian build 12.5 (Bookworm). What is the solution?

```
$ ./QIDISlicer_1.1.4_Linux.AppImage
/tmp/.mount_QIDISlVx7RSV/usr/bin/bin/qidi-slicer: 
 error while loading shared libraries: libwebkit2gtk-4.0.so.37: 
  cannot open shared object file: No such file or directory
```

QIDITECH, 2024-07-01:
Yes, some linux systems don't come with libwebkit2gtk, so you'll need to install it manually.
sudo apt-get install libwebkit2gtk-4.0-dev

closed 2024-07-01 by QIDITECH

ChrisJones79, 2024-07-01:
Ok, thanks. What if 4.0-dev is unavailable, is a newer version ok?

kasunch, 2024-07-31:
Some of the latest distributions such as Ubuntu 24.04 upgraded the libwebkit2gtk to 4.1 and do not ship 4.0 any longer.  I solved the problem by extracting the appimage and creating a symbolic link as per the instructions in https://github.com/bambulab/BambuStudio/issues/3973#issuecomment-2097651206.

```bash
./QIDISlicer_<version>_Linux.AppImage --appimage-extract
ln -s /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 squashfs-root/usr/bin/bin/libwebkit2gtk-4.0.so.37
ln -s /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 squashfs-root/usr/bin/bin/libjavascriptcoregtk-4.0.so.18
``` 

You can run slicer using the following.
```bash
./squashfs-root/AppRun
```

ChrisJones79, 2024-08-01:
> instructions in bambulab/BambuStudio#3973 (comment)

Thanks a lot! I am not savvy on dealing with app images, I didn't know you
could fix them like that.