QIDITECH/QIDISlicer #81: Qidislicer 1.1.5 LinuxAppimage segmentation fault

joaocheri, 2024-07-04:
On Ubuntu, updating QIDI slicer to latest version,
I cannot run QIDI slicer anymore (even not when going back to version 1.1.4). Iḿ getting following error:
$ ./QIDISlicer_1.1.5_Linux.AppImage 

(qidi-slicer:15577): Gtk-CRITICAL **: 20:36:15.775: gtk_window_resize: assertion 'height > 0' failed

(qidi-slicer:15577): GLib-GObject-WARNING **: 20:36:15.792: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:15577): Gtk-CRITICAL **: 20:36:15.792: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:15577): GLib-GObject-WARNING **: 20:36:15.794: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:15577): Gtk-CRITICAL **: 20:36:15.794: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:15577): GLib-GObject-WARNING **: 20:36:15.798: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:15577): Gtk-CRITICAL **: 20:36:15.798: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:15577): GLib-GObject-WARNING **: 20:36:15.800: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:15577): Gtk-CRITICAL **: 20:36:15.800: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:15577): GLib-GObject-WARNING **: 20:36:15.806: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:15577): Gtk-CRITICAL **: 20:36:15.806: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:15577): GLib-GObject-WARNING **: 20:36:15.807: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:15577): Gtk-CRITICAL **: 20:36:15.807: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
No elefant_foot_compensation_layers in ConfigOptionsGroup config.
No precise_z_height in ConfigOptionsGroup config.
No precise_outer_wall in ConfigOptionsGroup config.
Segmentation fault (core dumped)

What could be the issue here?
Kind regards

QIDITECH, 2024-07-05:
Deleting the profile should solve the problem, but please make a backup before deleting it and provide it to us so we can figure out the problem.

j0hax, 2024-07-05:
Hello, I can also confirm that the AppImage version of QIDISlicer 1.1.5 segfaults on Fedora. Deleting or renaming `~/.config/QIDISlicer` and using version 1.1.4 works for now.

joaocheri, 2024-07-06:
Hi, I also can confirm that after deleting the profile, version 1.1.4 works.
I created a backup with 'tar zchvf QIDISlicer_profile.tgz .config/QIDISlicer', but when I try to attach it, it mentions that file type is not supported, however it says .TGZ is supported. Also renaming with capitlal TGZ, it doesn't work.

simmonsm, 2024-07-06:
1.1.5 segfaults for me as well on Mint. System details:

```
System:    Kernel: 5.4.0-186-generic x86_64 bits: 64 compiler: gcc v: 9.4.0 
           Desktop: Cinnamon 4.6.7 wm: muffin dm: LightDM Distro: Linux Mint 20 Ulyana 
           base: Ubuntu 20.04 focal 
Machine:   Type: Desktop System: 
           Mobo: Gigabyte model: Z270P-D3-CF v: x.x serial: <filter> UEFI: American Megatrends 
           v: F1 date: 11/29/2016 
CPU:       Topology: Quad Core model: Intel Core i5-7600K bits: 64 type: MCP arch: Kaby Lake 
           rev: 9 L2 cache: 6144 KiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 30399 
           Speed: 4085 MHz min/max: 800/4200 MHz Core speeds (MHz): 1: 4053 2: 4046 3: 4042 
           4: 4101 
Graphics:  Device-1: Intel HD Graphics 630 vendor: Gigabyte driver: i915 v: kernel bus ID: 00:02.0 
           chip ID: 8086:5912 
           Display: x11 server: X.Org 1.20.13 driver: modesetting unloaded: fbdev,vesa 
           resolution: 1920x1080~60Hz 
           OpenGL: renderer: Mesa Intel HD Graphics 630 (KBL GT2) v: 4.6 Mesa 21.2.6 
           direct render: Yes 

```
Running appimage gave:
```

rm -rf ~/.config/QIDISlicer/
./X-CF$ ./QIDISlicer_1.1.5_Linux.AppImage 

(qidi-slicer:1942973): Gtk-CRITICAL **: 12:17:44.085: gtk_window_resize: assertion 'height > 0' failed

(qidi-slicer:1942973): GLib-GObject-WARNING **: 12:17:44.099: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:1942973): Gtk-CRITICAL **: 12:17:44.099: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:1942973): GLib-GObject-WARNING **: 12:17:44.100: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:1942973): Gtk-CRITICAL **: 12:17:44.100: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:1942973): GLib-GObject-WARNING **: 12:17:44.103: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:1942973): Gtk-CRITICAL **: 12:17:44.103: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:1942973): GLib-GObject-WARNING **: 12:17:44.104: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:1942973): Gtk-CRITICAL **: 12:17:44.104: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:1942973): GLib-GObject-WARNING **: 12:17:44.109: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:1942973): Gtk-CRITICAL **: 12:17:44.109: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(qidi-slicer:1942973): GLib-GObject-WARNING **: 12:17:44.110: invalid cast from 'wxPizza' to 'GtkCellLayout'

(qidi-slicer:1942973): Gtk-CRITICAL **: 12:17:44.110: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
No elefant_foot_compensation_layers in ConfigOptionsGroup config.
No precise_z_height in ConfigOptionsGroup config.
No precise_outer_wall in ConfigOptionsGroup config.
Segmentation fault (core dumped)
```
I then removed the ~/.config/QIDISlicer folder and then ran version 1.1.4 and (apart from the GTK errors and warnings) it started OK.

However, since I have an X-CF Pro it turns out that this software doesn't include that machine's configuration option at startup so I'm done with it.

QIDITECH, 2024-07-08:
We have fixed this bug, please update the software to the latest version.

closed 2024-07-08 by QIDITECH