QIDITECH/QIDISlicer #84: High CPU usage on all tabs except "Device"

QuestionablyKosher, 2024-07-19:
In QIDISlicer-1.1.4 there is relatively high (10%) CPU usage on all tabs except for "Device".

I saw #82 mentioning it due to "the printer being offline while the software continues to obtain the printer status". But, the printer is current online, accessible via WIFI, and even printing.

When the "Device" tab is open the CPU drops to just able 0%.

The expectation would be the application is accessing the printer status the same amount, regardless of the tab

I can provide you anything you may need - screenshots, videos, etc.

QIDITECH, 2024-07-20:
Thank you for your feedback.
Please provide a video to help us understand the problem more clearly. Please record a complete video of CPU changes from software startup to switching between the various tabs.

QuestionablyKosher, 2024-07-20:
https://github.com/user-attachments/assets/86d06bea-a2d1-4963-874b-268325a3e802

Above I've attached a video moving between the tabs. This was almost right after startup.

Please let me know if you need anything else

QuestionablyKosher, 2024-07-21:
https://github.com/user-attachments/assets/579cddbc-1b8c-4469-ba39-ba99719c96dd

After doing some more investigation, it appears that on initial application launch, the CPU usage remains at 0%. It is not until navigating to the "Device" tab for the first time that the issue appears.

QIDITECH, 2024-07-22:
Your question should belong to #47, not #82. Please try to update the software to 1.1.5.

QuestionablyKosher, 2024-07-22:
Apologies for referencing the incorrect issue.

It appears 1.1.5 fixed this issue. Should have upgraded earlier.

Thank you for the resolution

closed 2024-07-22 by QuestionablyKosher