QIDITECH/QIDISlicer #86: Qidi Slicer Appimage for linux fails to Launch

templaricknight, 2024-08-19:
I am running Nixos and have tried multiple ways to run this appimage, but everyway it fails.. There seems to be an issue with the dependency with webkit2gtk

QIDITECH, 2024-08-19:
Bundling WebKit into the AppImage is difficult because it leads to various issues with its dependencies. So far we did not manage to create an AppImage that would reliably run on all relevant distros.

QIDITECH, 2024-08-19:
Some linux systems don't come with libwebkit2gtk, so you'll need to install it manually.
```
sudo apt-get install libwebkit2gtk-4.0-dev
```
Or, some of the latest distributions such as Ubuntu 24.04 upgraded the libwebkit2gtk to 4.1 and do not ship 4.0 any longer. Please refer to #78.

QIDITECH, 2024-11-20:
We have uploaded two Linux versions compatible with libwebkit2gtk-4.0 and libwebkit2gtk-4.1 in the latest test version, please download and test.
https://github.com/QIDITECH/QIDISlicer/releases
If there are no problems, I will close the issue after a week.