QIDITECH/QIDISlicer #88: No connected 3D Printer and no "send to printer" option

Anercomp, 2024-09-15:
After upgrading to 1.5.1 my printer is not automatically detected anymore. When opening the printer in "device" manually it shows up fine with all stats and camera feed and saying standby. Any help is appreciated, as there is no option displayed to send files to the printer . Thanks!
[image]

QIDITECH, 2024-09-16:
Have you ever tried removing a device and adding it again?

r00tdownx, 2024-09-23:
I have same problem!  I had to roll back to 1.15
[image]
[image]

r00tdownx, 2024-09-23:
Error uploading to print host:
HTTP 500: {"error": {"code": 500, "message": "Internal Server Error", "traceback": "Traceback (most recent call last):\n\n File \"/home/mks/moonraker-env/lib/python3.7/shutil.py\", line 563, in move\n os.rename(src, real_dst)\n\nOSError: [Errno 18] Invalid cross-device link: '/tmp/moonraker.upload-464.mru' -> '/home/mks/gcode_files/Handle 008(20mm Screw).gcode'\n\n\nDuring handling of the above exception, another exception occurred:\n\n\nTraceback (most recent call last):\n\n File \"/home/mks/moonraker/moonraker/components/file_manager/file_manager.py\", line 673, in _process_uploaded_file\n upload_info['dest_path'])\n\n File \"/home/mks/moonraker-env/lib/python3.7/shutil.py\", line 577, in move\n copy_function(src, real_dst)\n\n File \"/home/mks/moonraker-env/lib/python3.7/shutil.py\", line 263, in copy2\n copyfile(src, dst, follow_symlinks=follow_symlinks)\n\n File \"/home/mks/moonraker-env/lib/python3.7/shutil.py\", line 122, in copyfile\n copyfileobj(fsrc, fdst)\n\n File \"/home/mks/moonraker-env/lib/python3.7/shutil.py\", line 82, in copyfileobj\n fdst.write(buf)\n\nOSError: [Errno 28] No space left on device\n\n\nDuring handling of the above exception, another exception occurred:\n\n\nTraceback (most recent call last):\n\n File \"/home/mks/moonraker/moonraker/app.py\", line 879, in post\n result = await self.file_manager.finalize_upload(form_args)\n\n File \"/home/mks/moonraker/moonraker/components/file_manager/file_manager.py\", line 532, in finalize_upload\n result = await self._finish_gcode_upload(upload_info)\n\n File \"/home/mks/moonraker/moonraker/components/file_manager/file_manager.py\", line 604, in _finish_gcode_upload\n finfo = await self._process_uploaded_file(upload_info)\n\n File

r00tdownx, 2024-09-24:
Sorry my problem was printer was hard drive was full but was giving network time out error!

QIDITECH, 2024-09-24:
As for the bug report, we have some problems with our judgment, which will be fixed in the next version.

QIDITECH, 2024-11-20:
The issue should have been fixed in the new version, if it still exists, I will open the issue again.

closed 2024-11-20 by QIDITECH