QIDITECH/QIDISlicer #89: Slicer 1.1.5 crashes when slicing

max-allan, 2024-09-22:
[cones.zip](https://github.com/user-attachments/files/17089232/cones.zip)

Nearly every time I try to slice this file (2 cones downloaded from thingiverse on the same print) Qidi slicer crashes.

I can slice each one individually without any trouble.

Removing one cone from the bed and readding it, I can sometime slice, but then if I change print settings and reslice: Crash.

If I make one non-printable and slice: Crash.

Running latest version 1.1.5

QIDITECH, 2024-09-23:
I'm sorry, I can't reproduce your question.
Check whether it is caused by insufficient memory or other reasons.
Providing more detailed videos or pictures might lead me to something new.

max-allan, 2024-09-23:
How much memory did you need to slice it? I have 32G, should be sufficient right? 

Which "other reasons" could there be?

max-allan, 2024-09-23:
https://github.com/user-attachments/assets/ba3d1c2d-1b56-4c06-a23d-8eed439ebf94

Selecting a 0.16mm or 0.2mm print settings seems to contribute. I have checked for configuration updates and app updates and everything is now running the latest version.

Choosing 0.12mm profile and it works fine.

max-allan, 2024-09-23:
I can add other objects to the build, just not those 2 cones...

QIDITECH, 2024-09-24:
It was very strange, and I was not sure what the problem was. I will try it on more computers and try to reproduce your problem.
Please try the following to avoid the crash, and let me know the results:
1. Move or rotate the models
2. Delete one of the models and slice one model at a time
3. Repair the model before slicing

QIDITECH, 2024-11-20:
We fixed the issue in the latest version, but I couldn't reproduce it, so I'm not sure if the fix was successful. Please download the latest version and test it.
https://github.com/QIDITECH/QIDISlicer/releases
If there are no problems, I will close the issue after a week.