QIDITECH/QIDISlicer #90: "Send to Printer" dialog box unreadable

y-e-r, 2024-09-24:
QIDISlicer 1.1.6 AppImage, when system gtk theme is dark the "Send to printer" dialog box is unreadable. Dark text on dark background, and light text on light background. 
[image]

QIDITECH, 2024-09-24:
OK, we will try to fix it in next version.

wabbitguy, 2024-09-26:
@QIDITECH Mac_OSX86 has the same issue...

No idea what's supposed to be in those check boxes where the big white squares are. Be nice if you not only fixed it but explained what is "supposed" to be showing.

[image]

QIDITECH, 2024-09-27:
In light mode, he is normal, only in dark mode will appear this way.
We will fix this issue in a future release.

wabbitguy, 2024-09-27:
@QIDITECH thanks for the info. Least I know what the text is supposed to say now. Thanks.

wabbitguy, 2024-09-29:
@QIDITECH update QIDI Slicer, now version 1.1.7 dark mode dialog box still not fixed.

QIDITECH, 2024-09-30:
Yes, due to time constraints, we prioritized fixing some other issues, and this bug is still being fixed.

jonXmack, 2024-10-17:
[image]

Thought I'd upload this so you can see what it actually says!

QIDITECH, 2024-11-20:
We have fixed this issue in the latest version.
https://github.com/QIDITECH/QIDISlicer/releases

closed 2024-11-20 by QIDITECH