QIDITECH/QIDISlicer #92: QIDISlcer 1.1.7 not starting on Linux Mint 22

DanielH1987, 2024-09-29:
The following error occurs when trying to run Version 1.1.7 on Linux Mint 22 (Wilma):

error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

The Version of the libwebkit2gtk supplied with Linux Mint 22 is: libwebkit2gtk-4.1-0
Installing the older, required version fails due to dependencies that cannot be easily resolved.

QIDITECH, 2024-09-30:
Yes, you need install libwebkit2gtk-4.0-37 first.
```
sudo apt install libwebkit2gtk-4.0-37
```
We will try to make it compatible with libwebkit2gtk-4.1, but it will take some time.

outsideobserver, 2024-11-16:
> Yes, you need install libwebkit2gtk-4.0-37 first.
> 
> ```
> sudo apt install libwebkit2gtk-4.0-37
> ```
> 
> We will try to make it compatible with libwebkit2gtk-4.1, but it will take some time.

Hello,

Also using Linux Mint, this doesn't work. Terminal throws errors:

E: Unable to locate package libwebkit2gtk-4.0-37
E: Couldn't find any package by glob 'libwebkit2gtk-4.0-37'

How can I run QidiSlicer in Linux Mint 22 for the time being?

QIDITECH, 2024-11-19:
We will release a test version compatible with libwebkit2gtk-4.1 in the near future, please wait a little longer. Or use another operating system to run the software.

QIDITECH, 2024-11-20:
We have uploaded two Linux versions compatible with libwebkit2gtk-4.0 and libwebkit2gtk-4.1 in the latest test version, please download and test.
https://github.com/QIDITECH/QIDISlicer/releases
If there are no problems, I will close the issue after a week.

outsideobserver, 2024-11-20:
> We have uploaded two Linux versions compatible with libwebkit2gtk-4.0 and libwebkit2gtk-4.1 in the latest test version, please download and test. https://github.com/QIDITECH/QIDISlicer/releases If there are no problems, I will close the issue after a week.

Tested the 4.1 version and it runs great so far!

closed 2024-12-20 by wjyLearn