QIDITECH/QIDISlicer #94: Open QIDI Slicer updates in my preferred browser - not the Chrome-in-slicer browser

leckert123, 2024-10-12:
I do not want to download updates from within the slicer.  It opens a mini-browser of the Git page from within the slicer and always downloads to my PC download folder.  

1.  That is not where I want to download the file
2. There are no options in the slicer to change the download folder, or to have it ask me where I want it to be downloaded.
3.  Changes to the default download folder for the PC, having it ask before downloading, and location for MS Edge have zero affect - and even if it did, I'd not want to change the location for ALL programs

What I want:
4.  Have the updates open a link to my default Firefox browser, which is courteous enough to ask where I want to download the file, AND it remembers my download preferences as well.

QIDITECH, 2024-10-14:
You can select the download address in the Preferences.
[image]

leckert123, 2024-10-14:
That is selected as shown. Unfortunately the Slicer does not follow this setting and simply dumps it into the local downloads folder.

[image]

QIDITECH, 2024-10-15:
I'm sorry I misunderstood you a little bit.
When you start the software and automatically check updates, the next button does not appear, so you can only download the installation package from the web. Manually check updates when the next button can be displayed normally, and select the download address.
We will fix this feature in a future release.

QIDITECH, 2024-11-20:
We have fixed this issue in the latest version.
https://github.com/QIDITECH/QIDISlicer/releases

closed 2024-11-20 by QIDITECH