QIDITECH/QIDISlicer #97: Seam gap seems to do nothing

KoeSource, 2024-10-26:
It looks as if the seam gap has no effect.
Whether I set 0% or 15% makes no difference in reality. A smaller gap is displayed in the slicer simulation.

Can anyone confirm this?

QIDITECH, 2024-10-28:
0.4 * 15% = 0.06mm, which is almost invisible to the naked eye, maybe you can set it larger to observe the effect

KoeSource, 2024-10-29:
How do i get those seam gaps ? Any idea?
something must be completely wrong...

outer diameter: 10mm
eSun PETG+HS, 0.8mm Nozzle, see zip for more info.

[image]
[qidi.zip](https://github.com/user-attachments/files/17563183/qidi.zip)

The only way I could correct it:
[image]

Gradbb, 2024-10-30:
I noticed that you have set the pullback distance to 1. Please set the pullback distance to 3mm or set it to the default value, which will effectively improve your seam.

KoeSource, 2024-10-30:
> 1. Please set the pullback distance to 3mm or set it to the default value, which will effectively improve your seam.

1. It doesn't change anyhting about the seam gap
2. It creates an other issue -> blobs and ziz, the surface ist really ugly
If the retraction is over 1.5mm, i hear a small explosion inside the nozzle every retraction it does.

[image]

[image]

Gradbb, 2024-10-31:
> > 1. Please set the pullback distance to 3mm or set it to the default value, which will effectively improve your seam.
> 
> 1. It doesn't change anyhting about the seam gap
> 2. It creates an other issue -> blobs and ziz, the surface ist really ugly
>    If the retraction is over 1.5mm, i hear a small explosion inside the nozzle every retraction it does.
> 
> The following pictures are printed with same settings but retraction on default:
> 
> ![grafik](https://private-user-images.githubusercontent.com/4213418/381657759-0a0e5718-0f63-4c75-9c74-fde783f87d4a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAzNDIxMjgsIm5iZiI6MTczMDM0MTgyOCwicGF0aCI6Ii80MjEzNDE4LzM4MTY1Nzc1OS0wYTBlNTcxOC0wZjYzLTRjNzUtOWM3NC1mZGU3ODNmODdkNGEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAzMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMzFUMDIzMDI4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YWNlNTYwOTdlZDNiZWZkOWQ4YTljNzYwMTEyMDVjMmYzNjQzODI4MGYyZDUyMGRhYjdlZGQ4MWNkMGI0YmMxMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ._ow_-imChs_ZbuvpbTNrWBadoqQxPpWxrrbsEXyBa58)
> 
> ![grafik](https://private-user-images.githubusercontent.com/4213418/381657895-f8dc3486-d176-43cf-b87c-5f206e7ff2f9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAzNDIxMjgsIm5iZiI6MTczMDM0MTgyOCwicGF0aCI6Ii80MjEzNDE4LzM4MTY1Nzg5NS1mOGRjMzQ4Ni1kMTc2LTQzY2YtYjg3Yy01ZjIwNmU3ZmYyZjkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAzMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMzFUMDIzMDI4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTAyMjIyMjA2MDdkZDBiYjFjNDc4OWJlYzQ5OTVmNTZjMzJiY2Y4ZjZhM2VjYjkyMmQ2MzFhYWJhNDU5YTJjYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.3A-kKbGjgDxLIFBd80KOe-lqtrdNF6N5zRlRa0JDDto)

[image]

I have tested your file again, and you can also improve your seams by modifying the 'Deretraction extra length'. This is the effect I have achieved, and my value is 0.4mm. This value should not be set too high as it may cause excessive surface extrusion.
[image]

KoeSource, 2024-11-11:
This leads to over-extrusion at the end.
It only leads to a slight improvement in the seam gap.
I only want to reduce or overlap the gap.
Is there no better solution?

[image]

Gradbb, 2024-11-12:
Hello, this is my printed result.
[image]
I made the following adjustments on your 3mf.
[image]
[image]
[image]
In addition, for nozzles with a diameter of 0.8mm, this situation is usually caused by overflow. You can also try lowering the printing temperature to improve this situation.

KoeSource, 2024-11-27:
Thank you very much for your help.
This is pretty good for me.
I printed those standoffs with your settings at 230°C:

[image]

closed 2024-11-28 by QIDITECH