QIDITECH/QIDISlicer #98: QIDI Slicer 1.1.7 possible memory allocation problem

GusMcN, 2024-10-30:
[Mandala.zip](https://github.com/user-attachments/files/17571737/Mandala.zip)
The STL file in the attached zip file fails to slice and crashes QIDI Slicer with an access violation - see 100%.jpg.
If scaled to 50% then it will slice, but at 60% it will fail with a 'bad allocation' error - see 60%.jpg.
My machine has 64 GBytes of RAM (see  SystemInfo.jpg) so I'm unlikely to have run out of memory - see also the Process Explorer histories which indicate no strain on resources.
I have previously sliced and printed this file on an Ender 3v2 using Cura slicer so was not expecting any difficulties to arise.

QIDITECH, 2024-10-31:
After testing, I found that the error was caused by the parameter 'Only one wall on top surfaces'.
We will fix it in the next version, please turn this parameter off temporarily.

QIDITECH, 2024-11-20:
We have fixed this issue in the latest version.
https://github.com/QIDITECH/QIDISlicer/releases

closed 2024-11-20 by QIDITECH