QIDITECH/QIDISlicer #99: Variable Temperature & Flow Control

QidiBoi, 2024-10-31:
Hello QidiTech - 

I was wondering/hoping that your team may already be working on incorporating variable temperature & flow control g-codes into QidiStudio - see links below.   Please advise.   

Variable Temperature 3D Printing – The FUTURE of 3D Printing?
https://www.youtube.com/watch?v=P6Y8uUPd3yg&ab_channel=PRINTINGPERSPECTIVE   

SB53 G-Code Flow/Temperature Controller V1.1
https://github.com/sb53systems/G-Code-Flow-Temperature-Controller/releases/tag/V1.1

Gradbb, 2024-10-31:
Hello, currently there is no such feature. If your requirement is to make the outer wall shine the same, you can enable "Don't slow down outer walls".
[image]

QidiBoi, 2024-11-01:
I'm not finding the variable you mentioned above - where is this located?  I'm using qidistudio 1.9.7.53

Gradbb, 2024-11-02:
[image]
It is located on the filaments settings page.
If you have any questions about QIDI Studio, you can ask them on the [page.](https://github.com/QIDITECH/QIDIStudio/issues)

closed 2024-11-06 by QIDITECH