QIDITECH/QIDISlicer #100: updated to 1.1.7

sweetgroom, 2024-11-08:
I update Qidislicer to 1.1.7 and now I can't "send to printer" and it automatically slices and generates gcode without having to click "slice"

QIDITECH, 2024-11-09:
Automatic slicing is because you enabled background processing.
And about to can't "send to printer", I think it is because you did not choose the corresponding physical printer.
[image]

sweetgroom, 2024-11-09:
Hi
Thank you for your response. I was connected to the corresponding physical printer I think because I was getting the printers info in Fluidd on my laptop.  It was connected to the same up address as the WiFi

Lisa

image.png (view on web)<https://github.com/user-attachments/assets/2d377b80-0615-4d55-9d85-51ea750c15b4>

QIDITECH, 2024-11-12:
Sorry, I don't understand what you mean. Could you please provide videos or pictures to explain it.

closed 2024-11-20 by QIDITECH